JoernL/LimiTTer #1: LIMITTER

pedrojrr, 2016-04-16:
Hola mi nombre es Pedro y soy padre de un niño Diabetico que usa el freestyle libre. Primero quisiera darle mi enhorabuena por el desarrollo del proyecto ya que llevábamos tiempo buscando algo parecido.  Estoy deseando que comparta su diseño para poder probarlo en mi hijo sobre todo por la utilidad de guardián nocturno.  Gracias por su tiempo y dedicación estoy convencido que seremos muchos los padres que se lo agradeceremos. Espero tener pronto noticias suyas

holmberg81, 2016-04-16:
English would be more easier to understand.

berniemcintosh, 2016-04-21:
I speak a little spanish - here is a rough translation:

Hi, my name is Pedro and I am the father of a Diabetic boy that uses the Freestyle Libre. First I want to give my congratulations for developing the project since we have been waiting some time for something like this.  I want you to share the design for us to be able to use it with my boy, above all, for the overnight watch. Thanks for your time and dedication. I am convinced that there will be many of us parents that thank you. I hope we have more updates from you soon.

JoernL, 2016-04-21:
@berniemcintosh , you are welcome.

JoseRobotnik, 2016-04-24:
Congratulations for developing such as great project. 
I ordered some of the components of the limitter in aliexpress.com, except the BM019 NFC2Serial. Is there any other NFC2serial compatible dongle to replace it? 
Many thanks

JoernL, 2016-04-24:
I dont't know any other working module.

libxmike, 2016-04-26:
Joern, is there any trick to pair with Android?
I have Nexus with Android 6 and can't pair, not with the default pin or any other, 
can't see it in xDrip, any ideas?

To see the name "Limitter" in Settings-> BT, I had to change the code to:
ble_Serial.write("AT+NAMELimitter");

JoernL, 2016-04-26:
Settings -> Apps -> xdrip -> permissions. Enable location services.

tim2000s, 2016-04-26:
Thanks Mike. I'll try that. I can see the HMSoft device in xDrip without that mod and it pairs, although it didn't transmit any data. Do you have the correct baud rate set?

birdfly, 2016-04-26:
Hello,I  am a T1D for six years and user of a free style libre,I just  wonder that LimiTTer(the hardware )could interpretate the Tag-data  or transmit  the raw data to xDrip directly.For I have used several libre apps ,the value they read are different from the official reader.

tim2000s, 2016-04-26:
Okay, my progress so far is that I've built the LimiTTer, and that all the components communicate. I can confirm the Arduino is speaking to the HM10 as, with the earlier tweak, I was able to get the module renamed correctly. 

The assembly then connects to xDrip on the phone successfully. 

At the five minute intervals, the NFC card comes to life and I am assuming it scans, and I see the green LED flash on the Arduino. What I don't get is anything in xDrip. Nothing is showing up. 

I welcome any ideas on what I might need to change. The baud rate is set to 9600 as is the default on my HM10. Just not picking anything up and transmitting it.

JoernL, 2016-04-26:
You are not able to start a sensor in xdrip? The 2 start readings don't come in?

tim2000s, 2016-04-27:
That's right. Whilst xDrip sees the Bluetooth and connects, and I see the BM19 trigger, I'm not seeing anything come in on the two start readings. I'm running android 5.1.

JoernL, 2016-04-27:
The Hardware Data Source in xDrip settings must be set to "Bluetooth Wixel". Maybe you have changed this?

JoernL, 2016-04-27:
@birdfly  as yet, it only reads the last saved reading from the sensor and send it to xDrip, which modify the raw value through a slope and an intercept and then displays it at it's graph. I'm working on it now to find out how the original reader can precalculate the readings when glucose rising fast? At this moment the raw reading is the nearly the same but the LibreReader shows a very higher value. Maybe a hidden dynamically slope multiplier in the tag? Didn't find it yet.

berniemcintosh, 2016-04-27:
Joern,

I commend your work and find it most interesting.  My own feeling is that I think I prefer to see a raw reading from the Libre, with a simple adjustment to put it into mmol/ml. I think the need for predictive algorithms is perhaps not as great as the stress that some put on them.  

My take on it is that knowledge and context are the most important thing. I know that the interstitial glucose level lags behind BG as measured with a finger stick. This in itself lags behind venous level. Just as we are all used to the lag in BG fingerstick readings, maybe we should all get used to the lag in interstitial readings.

If I see that the Libre raw reading is fairly constant, then I can have good confidence that it reflects the reality. Constantly high, maybe I need to take a small amount of insulin. Level of 4.0 for half an hour and I'm about to exercise then maybe a little glucose is in order.

If the Libre reading is rising fast, I can judge for myself if there is a single reading that is a little out of step or if there is a trend. If there is a trend, then I can see how sharp the trend is, and for how long it has been there. I can look a the picture and let my brain fathom it  out using its massive parallel processing power and pattern recognition abilities in order to understand what is happening. An actual, predicted level that matches a BG fingerstick I don't believe is my ultimate objective. I can guess at what it might be and it will probably be as correct as any complex multiple derivative based algorithm. 

Perhaps the trouble with the algorithm is you don't quite have a grasp of what the algorithm has done. At least with raw data I can use my experience.

Likewise when BG is dropping. I see that the raw data is plunging from 13 to 10 to 7 to 5 in 25 minutes. I think I'm not concerned that I don't have a predictive current value equivalent to BG fingerstick. I want  some glucose please, NOW. Even though it says 5.  My experience tells me what is going to happen next. Not that I have such dramatic incidents too often, but it can happen.

So adjustment of data with slope and intercept yes, but predictive algorithm like the built in one - kinda looks good but not necessary. 

As you know occasionally the Libre refuses to give you a reading.  Wouldn't you just love to know what the raw values are that led to that decision. Well I guess we do now with the BM019 but I've yet to see it for myself.

Sorry - not trying to put you off - its good to have a predictive algorithm too. The more info the better. J

Cheers

Bernie

tim2000s, 2016-04-27:
Definitely set to the correct source. That's what is so odd.

SugarBots, 2016-04-27:
Tim- have you tried using a Bluetooth tool like Adafruit Bluefruit LE or BLE ( I use the ones for iOS) to confirm the BLE packets? 
And have you successfully paired the BLE in both android settings and xDrip? I haven't had issues with getting the data to display on xDrip, just that the mmol conversion needed to be added in.

Joern- which algorithm are you using to retrieve the current BG value? Is it debased on LiApp or Glimp? The Glimp calculation makes the process much clearer and provides the previous 15 values showing how trend can be predicted. Would be useful to be able to send BLE request to bm019 to "fill the gaps" and provide previous values too.

tim2000s, 2016-04-27:
I've not tried to pair it in android settings as experience with xDrip in the past was that you just paired from xDrip. That can easily be rectified though, just not this weekend!

JoernL, 2016-04-28:
@tim2000s, there is no need to pair it in Android settings.

@SugarBots, i only take the latest (recent) raw value and send it to xDrip's slope. I have a test version of xDrip installed to find out the best slope setting of that raw value. I have find out that a fixed slope of 1.15 works best by now. What about a second version of the LimiTTer sketch which pre-calculates more agressive based on the last 15minutes? So you can decide oneself's favour.

holmberg81, 2016-04-28:
Can I use this to get it smaller. :)
Only missing nfc now. :)

Martin

holmberg81, 2016-04-28:
Forgot link.
http://m.ebay.co.uk/itm/Digispark-Kickstarter-Attiny85-USB-Development-Board-for-Arduino-BEST-/141672395445?nav=SEARCH

LibCGM, 2016-04-28:
@holmberg81 I have one of this, will see if the code will fit the memory.

LibCGM, 2016-04-28:
@holmberg81 but I there is not enough GPIO's

SandraK82, 2016-04-28:
Not Enough IO? How many do you need? You can Drive the bm019 via uart like the hm10.
An example lib can be found here https://github.com/SandraK82/bm019-library/blob/master/mbed/ble-bm019-uart/bm019.cpp <https://github.com/SandraK82/bm019-library/blob/master/mbed/ble-bm019-uart/bm019.cpp> which is not yet Ported to arduino.

sean75, 2016-04-30:
Hello people,
I write from Italy and I have a boy ten years old with diabetes.
He use freestylelibre and I have many interest to build LIMITTER.
Thanks.

JoernL, 2016-05-02:
@sean75 , you're welcome.

@all, the current code produces moderate readings. The next update will be more aggresive and faster in rising up and down - like the original reader.

LibCGM, 2016-05-02:
for some reason it worked just once, xDrip receiving data from LimiTTer, do I need to stop/start if I test with another/different old sensor?

Will this work with HM-06 and BLE off?

JoernL, 2016-05-02:
@LibCGM , do you use Android 4.x and the BLE OFF circuit? BLE OFF only works with Android > 5.
You don't have to stop/start a new sensor in xDrip. The LimiTTer will scan every sensor. If you program new code to the Arduino, it's best to hit "forget device" in system status and make a new BTScan + Connection. Not necessary to stop the sensor in xDrip.
Yes, you can try it with your HM-06 and the BLE OFF circuit. Maybe you should increase sleep time a little bit for max power saving. But Android 5 or 6 only ;)

libxmike, 2016-05-02:
@JoernL I use Nexus 5 with Android 6, I will try today with HM-06, it is just BT 2.1.

I did all what you write with my current setting with HM-11, forget device, new BTScan etc, HM-11 ist connected but xDrip not receiving data, in console (on mac over USB) I see the BM019 reads the sensor.

libxmike, 2016-05-02:
still nothing, not with HM-06 or HM-11, in System Status says connected but noting received

jamesthurlow, 2016-05-02:
Nice work Sandrak82 - having the BM019 work on UART is great.. the Adafruit Feather uses the SPI bus for bluetooth connections and swapping which device uses the SPI is proving tricky.

SandraK82, 2016-05-02:
with an ble uart friend from adafruit you can even omit the needed power for the mcu. I have competed (still not made public) a firmware replacement for the ble friend to drive the whole setup from the ble Module.
My setup needs only Lipo, Charger, ble friend and bm019. Connect bm019 like ss0 to ground, vin to lipocharger, ground to lipo charger, din to tx ble friend, dout to rx ble, vdd to vin ble, ground ble to ground charger ( http://unendlichkeit.net/wordpress/wp-content/uploads/2016/05/IMG_1542-768x1024.jpg ).

tim2000s, 2016-05-02:
I get the same issue with an HM-10 attached. Everything is wired up correctly (I've checked that a couple of times) and the Arduino has been reflashed to make sure it's working properly. 

Wonder what the issue is.

tim2000s, 2016-05-02:
I was wondering whether the device in the following link might use the same firmware as Joernl's initial attempt but provide an all in one solution. 

https://www.coolcomponents.co.uk/blend-micro.html

SandraK82, 2016-05-02:
i think that should be possibel, maybe you have to adapt the commands to the ble Module and the pin definitions. Nordic provides an Arduino lib for the nrf8001 ble https://github.com/NordicSemiconductor/ble-sdk-arduino

libxmike, 2016-05-02:
@tim2000s it should work, ATmega32u4 uses Arduino Leonardo but the one you linked is not word 38 quid, Arduino Mini Pro + HM-11 about $10

tim2000s, 2016-05-02:
Update: For some reason after a restart, it is working. I'm getting data on xDrip from a live Libre sensor.

libxmike, 2016-05-02:
@tim2000s what did the magic? 
I slowly give up, my stick with my working Simblee solution and iOS app.

JoernL, 2016-05-02:
@tim2000s, i think it will. But the modules length is 43mm. I've chosen the pro-mini because its only 32mm. Every mm counts ;) It must be as small as possible cause you wear it every time.

JoernL, 2016-05-02:
@libxmike, tomorrow i will upload new code. Maybe this will fix your problem.

libxmike, 2016-05-03:
@JoernL thanks, will be great if it works, and thanks for your time you investing in it.

sean75, 2016-05-03:
Dear Sir,
I write from Italy, south Italy,and my son is a diabetic boy ten years old.
He use frestyle libre and I have a lot of interest for limitter.
Can you give me some information to build it?
Sorry for my bad english.
Thanks
--
Inviato da Libero Mail per Android martedì,

tim2000s, 2016-05-03:
I think it depends on how you wear it. If it's in a neoprene sleeve, fewer devices and lying next to the BM19 is fine.

I'm trying to avoid the box!

tim2000s, 2016-05-03:
Look forward to this!

LibCGM, 2016-05-03:
@tim2000s box is not an optimal solution, I designed and 3D printed one, it is too bulky
http://imgur.com/8nMsZuy
I may design one with round edges, shallow/flat and reassembles you arm - arc shape, the frame is 1-1.5mm thick

SandraK82, 2016-05-03:
we are exploring the option to use flexible boards embeded in Silikone for the bluereader. The only bad thing is, that firmware over the air has to work correct, or the Silikone must be openend. Loading will be wireless, probably qi std. Batteries might be two rechargable li-ion coins cells

LibCGM, 2016-05-03:
@SandraK82 Silicone my be the same problem as Abbott glue, maybe warped in a fabric to avoid Silicone - skin contact? but again it wont be on 24/7 like the Sensor so you skin can breath.
Silicone is ok if you want to have it on 24/7 w/o taking it off, Qi charging was also my consideration but in addition to micro USB in case you are not near a charger. It has to be moist but not waterproof, you don't want to swim or shower with it.

SandraK82, 2016-05-03:
Yes it might be, but so far it seems to be the best option we can think off. For the usecase as libre reader, the solution does indeed not be waterproof, however we are targeting a broader audience (like Industrie) with the bluereader. Therefore it needs to be resistente against Shock, fluids, chemicals and needs a a easy handling.
My current protoype setup is only warped in  insulation tape and needs a 5v breaboard Adapter for charging (currently no lipo charger with usb port, order does not arrive). I found fixing the the board with medical tape (Leukosilk) works best for me. I applied a curapor plaster over the sensor to further protect my skin ( i got burns from an unprotected contact of my boards once). I Experimente also with arm cases for mp3 players, bandages and a custom sawn armband but i was not able to lock the board over the sensor for a day without squeezing my arm to tight.

tim2000s, 2016-05-03:
I'd been looking at the flexible PCB options too. 

The issue I'm running into with all of this is it starts to remove the DIY aspect and we run into where liability lies. With the current xDrip kit, the user has to solder a component in place and do the software upload. 

Is an OTA software upload by the user enough to alleviate liability risks?

SandraK82, 2016-05-03:
As far as we have Look into the liability aspect we found that a generic device should have no liability or guranties besides from working safely within the specifications made by us. Our device does not support xdrip upload with the original firmware, not even Interpretation of the libre data, it just acts as an ble to nfc gateway for many different nfc card and tag types…
So if you decide to use an app which utilizes the bluereader library to Connect to an bluereader and then communicates with the libre sensor and shows an Interpretation of the data obtained, or if you opt for replacing the open source bluereader firmware with a own diy firmware which can talk directly to a xdrip, then this should be beyond the scope of our liability :) at least i hope so and got a correct understanding of the different national and international laws

LibCGM, 2016-05-03:
@SandraK82 here is my try to fix it with Velcro http://imgur.com/yWh2rkH , I can even sleep on it but its far from optimal or safe :) 
good way to experiment with flexible case is 3D print with e.g. NinjaFlex, just have to wait til I have parts for my Ultimaker 2 clone then can do some case experiments.

jolaca01, 2016-05-05:
Congrats on your development JoernL and everyone in this community for trying to improve your original design!! I'm also very interested in all your findings. Thanks again for your interest and work!!

libxmike, 2016-05-06:
@JoernL finally got it working with you new version of both xDrip and LimiTTer.ino, what it was is still a mystery :) Thanks!

JoernL, 2016-05-06:
@libxmike, fine :) Enjoy!

libxmike, 2016-05-07:
@JoernL is it possible to send the packets direct to NightScout/Azure DB over WiFI with ESP8266 instead over BT?
the only problem I see is the ESP8266 needs some power

JoernL, 2016-05-07:
Hmm ... i've got none experience with uploading the data to any database. All i know is the dexcom community always uses xDrip to upload their data - not directly the wixel. But i think it should be possible. But then the data must be parsed into an acceptable format like xml or something else.

libxmike, 2016-05-07:
@JoernL I found one problem with xDrip for LimiTTer, in shows LOW alarm when the BG high is, red LOW letters on the screen

JoernL, 2016-05-07:
Yes, i have already fixed that. Will upload new code next days ...thx!

Maik-dyk, 2016-05-08:
Hi everyone,
I ordered a BM019 but the circuit is out of stock and they can’t serve me until June. Do you know any shop that serves in Europe and has this item in stock?
I want to start testing the Limitter.
Thanks in advance.

jolaca01, 2016-05-08:
Hi, same problem here Maik-dyk!! Any idea where to get in a reasonable period of time(=in stock)  a BM019 in Europe?? Thanks!!

[110 more replies]