JoernL/LimiTTer #2: the xDrip experimal

birdfly, 2016-05-09:
Hello 

I  know that the xDrip is developed for Dexcom G4,and I find some codes to Interpretate the data from Wixel BT  devices ，but our Limitter is desighed for Freestyle Libre and the data from two devices may be different ,have you modified the codes for the data send from Limitter .

JoernL, 2016-05-09:
Yes, the LimiTTer code converts the value into a raw dexcom value by a simple multiplier.
Dexcom raw values are much higher.

birdfly, 2016-05-09:
so this function is accommplished in the hardware or the xDrip modified for Limitter

JoernL, 2016-05-09:
It's accomplished by the Arduino.

birdfly, 2016-05-09:
if I want develop an app which have more features than xDrip （for example add the insulin and food etc）,how can I receive the data from my Limitter

holmberg81, 2016-05-09:
Hi
some one now if LimiTTer works to Nexus 5x thinking to buy  one. :)

Doing the last part now off LimiTTer hope it works to the weken.

Martin

roppenheimer, 2016-05-13:
Can you post where you purchased the various components from. Most of them are no problem but I have been unable to locate a suitable box.

Thanks, I really appreciate the work you have been doing on this.

JoernL, 2016-05-14:
I've purchased this one:

(http://www.voelkner.de/products/290989/Miniatur-Kunststoff-Gehaeuse-1551qgy.html)

JoernL, 2016-05-14:
@birdfly, your app should receive a simple string with 3 values via BT. Then cast that values into an int, long, or what you like.

roppenheimer, 2016-05-14:
Thank you. I have now found a local supplier of this casing. Hammond Electronics manufacture a range http://www.hammondmfg.com/dwg9.htm

 

(http://www.voelkner.de/products/290989/Miniatur-Kunststoff-Gehaeuse-1551qgy.html)

jolaca01, 2016-05-16:
Hi Joernl, Thanks for the cases link!!. I really like your chosen case but I would prefer to leave the NFC reader protected inside the case (cutting it)  in more or less this way:
[image]

So, I guess for this, a longer case would be needed, as I don't have yet my components here, what do you think: 1) would the NFC reader be able to read if I close the case this way? and 2) 50x50x20 or 50x35x20?? PLease, let me know your thoughts!

holmberg81, 2016-05-16:
Hi I have in case. 47x45mm and 15mm
And batteri 250mA

jolaca01, 2016-05-16:
Hi Holmberg81 have you got a link to your case or could you share a picture? Are you planning to cut down the case in the same way??

holmberg81, 2016-05-16:
New box in 3 days. Send you new one then.

JoernL, 2016-05-16:
@jolaca01 , maybe with your design idea the case should be 3d printed. Then the front area could be solid where the BM019 sticks in at the bottom. The exterior of the tongue (BM019) can then be added with velcro tape that sticks on the sensor. But the wall thickness must be thin at the tongue. Good idea! :) Yes.

jolaca01, 2016-05-16:
Thanks JoernL!! Definitely it would be great to have it printed but I have ordered a pair of cases and cutting down and retouching one I'll try to get a first version sorted out. Then,  it would be nice to have a neat perfect dimensions  3d printed case. ;-)

EvertKlevert, 2016-05-23:
Is there anyone who could make me one? I'd buy it from you, for a nice price. I live in the Netherlands. (Oh and I'm certainly not that techy to make one by myself....)

jolaca01, 2016-05-25:
Hi @EvertKlevert, I'm still waiting for an NFC module to build the one for my daughter, I hope it will arrive this week. If everything goes well I wouldn't mind to build one for you but it's @JoernL designed so he may have something to say, I mean I  wouldn't want to bother anyone. ;-)

EvertKlevert, 2016-05-25:
Hie @jolaca01 Thanx for the offer! We're talking about the LimiTTer for freestyle, though?

(I get confused by sites talking about the xDrip and this LimiTTer. The xDrip is only Dexcom?)

I hope @JoernL will approve. This would be *the* most interesting way in making the Libre a CGM.

jolaca01, 2016-05-25:
@EvertKlevert  don't get confused, xDrip is also used by @JoernL LimiTTer to pass data via BT from the Arduino mini to your mobile and then from there to your  Nightscout site. And yes, the thing is to convert the FS Libre to something similar to a CGM device.

jolaca01, 2016-05-25:
To be clear, I mean xDrip is an app (on your mobile) that receives data (in this case form the LimiTTer) via BT and can send it via internet to your Nightscout site.

EvertKlevert, 2016-05-25:
Forgive me my confusion,, in mij search I found [this site](http://xdripkit.co.uk/) where a receiver is being offered, and it's called xdrip. So not an app, but a transmitter.

EvertKlevert, 2016-05-25:
In addition: the ^^^mentioned site which offers an xDrip hardware-set, this one is used for the dexcom g4. 
Mmm. In my quest for 'The Answer Of Knowing Everything On How To Make A CGM From Libre' I still haven't found the universal answer/guide.

JoernL, 2016-05-25:
@EvertKlevert , for dexcom systems "xdrip" mostly stands for both - hardware + app. The LimiTTer hardware is using the same "xdrip" app as the dexcom hardware (Wixel). In future ... xdrip app will be able to display data from a medtronic cgm system, too. One app for all systems. Only problem is, that xdrip app is Android only till now.

JoernL, 2016-05-25:
@jolaca01 , design like you want :) My design is only a reference design. With the instructions pdf you are able to make a copy of this.

jolaca01, 2016-07-08:
Hi, just wanted to thank JoernL again as his LimiTTer is working great for my daughter. I also wanted to share the final design I've used. Taking advantage of the fact that when you use Velcro the total height increases a little bit over the Libre sensor, instead of making a recess in the case, I think it's better to make a cut in the top of the case just for the NFC reader but keeping it inside. Thus, we can place half LimiTTer together with the Libre sensor, using less space. Here I attach a photo of it, though I've used a 50x50 case and it's a bit too large for my daughter's skinny arm. Nevertheless, I think it may work fine with a 40x40 case design.
[image]

NFCole, 2016-07-09:
I had this idea, too, but i wasn't sure if the nfc is strong enough... Looks great, well done ;)