JoernL/LimiTTer #5: Other NFC modules?

mattiastoyra, 2016-06-13:
Hello!
Really love this project of yours.
I try to build one for myself but ran into problems. I cant seem to get hands on a Solutions Cubed BM019 (out of stock whereever I look). I live in Sweden
This is my very first time trying to build something using Arduino and I hoped it would be as simple as swapping the BM019 for any other NFC/RFID module. 
I bought this out of E-bay http://www.ebay.com/itm/MFRC-522-RC522-RFID-Radiofrequency-IC-Card-Inducing-Sensor-Reader-for-Arduino-BE-/191371999374?hash=item2c8ea8f88e:g:igAAAOxyBjBTRMCs
and just tried to find the corresponding pins.

 But I cant get it to read Libre sensors, the bluetooth connects as it should though so I figured its the NFC module.

Do I have to rewrite all the code for this specific module or what am I doing wrong?
Can you recommend any other NFC module that are compatible with this project? 

Thank you!

JoernL, 2016-06-14:
This one does not work. You have to wait for the BM019. Don not worry ... the distributors have  noticed a huge increase of pre-orders and have ordered much more from SolutionsCubed

Verticoe, 2016-11-13:
Hi, 
first of all thanks for your wonderful work. Right now, I am in the process of building my own LimiTTer for my girlfriend. Most of the parts already arrived. Unfortunately, I made the same "mistake" than mattiastoyra, buying an RC522. 

So far, I was able to map the pins on the RC522 to the ports of the Arduino Pro Mini. Amongst other, I did the mapping with the help of this project here [https://github.com/AndreaBruno/OpenBtNFC/wiki](url). There is a nice sketch in the middle of the site. 
Anyway, due to your answer I hesitate to actually solder the parts together. I am rather inexperienced with Arduino and also taught myself how to solder just recently and I don't want to make experiments and want to follow your instructions as close as possible. Thus, I ordered the BM019 from Solutions Cubed.

Yet, I wonder if you made your own experiences with the RC522? Did you try the RC522 module before switching to the BM019? Why exactly does it not work? Is it something with the energy consumption?
Is there a chance to make it work by modifying the code of the LimiTTer.ino?  

Also, when the connecting the the BM019 to the Arduino Pro Mini, the Vin goes to 4,7 & 8. Did you just solder three pieces of wire together?

Thank you for your effort!

JoernL, 2016-11-13:
@Verticoe , RC522 does not support ISO 15693 NFC. On my current LimiTTer i have tested powering the BM019 by only one pin for a long time - and it works without any problems. So i can recommend this to save space in the case.

closed 2016-12-09 by JoernL

BabarRehman, 2017-12-16:
Hello, I have a problem, I am trying to Port your code to ESP32, my inventory command fails because the value returned is not 128, however it does detect the BM019 cause the values in rxbuffer change, I don't have experience in microcontrollers, can you help me with identifying the issue?