JoernL/LimiTTer #11: 'Glucose_Reading' was not declared in this scope

Jimlake, 2016-10-24:
when build the project. an error occurred with 'Glucose_Reading' was not declared in this scope
Where can we modify the code ? thanks.

JoernL, 2016-10-24:
Hmm ... take a look at line 369. Maybe a brace bug in your code?

float Glucose_Reading(unsigned int val) {
        int bitmask = 0x0FFF;
        return ((val & bitmask) / 8.5);
}

Jimlake, 2016-10-24:
It seems no obvious bug  int my code ,  I copy the above code again,but it still reprt the same error. 
my ardunio IDE version is the latest 1.6.12

Jimlake, 2016-10-24:
I move the function code above the function Inventory_Command(), compiled ok.

float Glucose_Reading(unsigned int val) {
        int bitmask = 0x0FFF;
        return ((val & bitmask) / 8.5);
}

void Inventory_Command() {

  digitalWrite(SSPin, LOW);
  SPI.transfer(0x00);  // SPI control byte to send command to CR95HF
  SPI.transfer(0x04);  // Send Receive CR95HF command
  SPI.transfer(0x03);  // length of data that follows is 0

closed 2016-12-09 by JoernL