JoernL/LimiTTer #13: Extend use of FreeStyle Libre sensor

renno-bih, 2017-01-11:
I'm glad to see projects like this. As this sensor can work only for 14 days it is very expensive for diabetics.
Is there any way to extend life of this sensor and still use it with LimiTTEr after 14 days. Looks like battery is not problem but Abbot forces you to buy another one.

BenY1, 2017-01-15:
I would also like to open one up and inspect it. I feel it may have some sort of internal clock that counts down once the tag is active to the reader. Its likely not possible to change anything. If its got an internal clock then short of modifying the clock on a nano level is impossible. They have us by the balls. I will strip one down soon to see how they made them.

JoernL, 2017-01-15:
I have opened (milling) an active sensor and tried to do a quick restart by pull and push the battery for a second - without success! It immediately stops working.

mantas-done, 2017-02-25:
In LibreLink app there is one interesting sentence:
Be aware that travelling into earlier time zones could shorten your sensor life if you are near the end of your sensor wear period.

So maybe the life can be extended by changing to different timezone?

Source:
https://www.librelink.com/support/getting-started/new-sensor

nicolasrk, 2017-09-29:
@mantas-done I think there u have a point. I didn't probe what happens if u move back ur date on the reader when is almost to finish the sensor time. Anyway I think the sensor should have some logic and flags to avoid this kind of situation. But is a good point to try a hack. 
I'am pretty sure the life time of the sensor is mostly defined by commercial reasons. (for sure there is tech reason like, oxidation of sensor, battery life time, etc, etc. But 14 days sounds really arbitrary IMO.

d-scott, 2017-10-24:
FYI some people have had success getting readings from the sensor after the 14 day period using the Glimp app on Android. Check the comments from this blog post:
http://www.diabettech.com/freestylelibre/freestylelibre-more-than-14-days-glimp-a-third-party-libre-reader-app-with-a-difference/

Perhaps the same can be done with LimiTTer?

darren060673, 2018-02-21:
Ok, so I'm new to github, but have been a very keen hardware tinkerer for many years. I have opened up several libre sensors and managed to download Ti datasheets for the family of proceesors. The chip has a reset pin than appears to be tied down to -ve with a resistor and capacitor of as yet unknown values. There is also a thermistor coupled to one of the processors ADC inputs. I am currently trying to figure out how to potentially reset the device.
Would be greatful of and software help, as I am purely a hardware guy.

darren060673, 2018-02-27:
I have just learned today that the USFDA have only approved the sensors for 10 day use, therefore it's got to be a software/firmware switch, most likely determined by the reader/starter. Has anyone managed to start a new sensor from a Non Abbot  App, like Glimp or Xdrip? apologies if I'm asking stupid questions, but I feel there has to be potential to  hard reset the processor.

valentinser, 2018-03-05:
Hello, just found this topic. I wanted to say that Glimp cannot start sensors. As far as I know, only the official Abbott FreeStyle Libre App or the Sensor Reader can start a new sensor.

waltsteinchen, 2018-03-07:
I read through the specs .. and imho the chip has a MSP430 as core and is completely programmed OverTheAir at startup (=enabling the sensor with the reader), imho it's a general purpose chip for reading data through NFC from TI.. nothing really special for Abott. I guess! that this is the reason that you cannot enable a sensor with the handy app because the risk that the app would be reverse engineered was considered too high. I also guess that the entire expiration programming is determined during that programming stage.  Thought a while why it is possible to read data from the sensor for >14 days, ~1 day or so... but the oscillator on the device is probably not very exact .. so they let it run for a little longer time and do the exact limitation in software. At least I would do that ;-) again .. all above is basically guessing.

SandraK82, 2018-03-07:
The chip is not a standard, but very similiar to a standard nfc chip from ti. The application is factory preloaded and the activation is just setting something via nfc. This is not only done by the official apps and already reverse engineered by the people who provide glimp-s.

Once activated the chip locks Write access and Starts the timer, this Runs usually 14 days and a half currently for me (European sensor). There are reports of longer running 
sensor from other countrie, but nothing concrete yet.

valentinser, 2018-05-01:
> All,
> My son has been using FreeStyle Libre sensors/reader that we purchase from Europe for the last 2 years.
> Yesterday we attempted to read/initiate a US purchased sensor with the European reader and I received an error indicating the sensor was not compatible (I'll try to post exact error later).
> FreeStyle Libre from Europe lasts 14 days and have a 1-hour warm-up instead of US version which lasts 10 days and has a 12 hour warm up period, which is ridiculous.

Hello, 
I think the 10 day period vs 14 day period might be related to different regulation and aplicable laws.

fernandezfj, 2018-05-01:
Hello,

I read part of the FDA approval, it seemed like Abbott was playing it safe and extended warm-up period and shortened the dates to be able to show even better data. I didn't see a restriction from the FDA per say...
Since I have both type of sensors I will try to take them appart and share pictures, but I don't anticipate finding any hardware differences. My guess is taht the software in the reader can detect the sensor it is authorized to work with and will dictate the warm-up and life of it. 
I finally got the prescription approved for my son in the US, and was very dissapointed to learn about the limitations.

glaucoqs, 2019-04-12:
Hi.. I'm from Brazil, and diabetic... Studying the sensor, the activation is in the moment when you open the new sensor and put 2 parts together... The pins of sensor, short the circuit, and let's battery energy starts the counter. Supposing the memory is for 8 hs and the sensor works about 14 days..  the counter is for the number of times the chip save the middle of 3 measures (1/min).. in total.. because in last day, with the app glimp, it's repeat in grafic..

I think the sensor can be used for much time... It's only a electrode sending and measuring the energy back.. to insert again in skin, can be used the needle in applicator...

cadcoke5, 2019-04-30:
The electronics are not the main issue that limit the lifespan of the sensor. Rather, it is the chemistry of the sensor and the body's own activities that affect the sensor filament that goes into the body.  

So, even if you succeed in getting the hardware to permit it to run longer than 14 days, the accuracy will quickly decline and eventually be  generating useless information.  And if you are going to medicate yourself based on useless information, you might be creating a very bad situation.

amrdys, 2019-05-15:
> The electronics are not the main issue that limit the lifespan of the sensor. Rather, it is the chemistry of the sensor and the body's own activities that affect the sensor filament that goes into [...]

At some point they say it will cause skin irritation If kept for longer periods.

IMO there should be a way to change parts of the sensor which can make things worth with continuous use (like the needle) and not all of it. I'm pretty sure the chip, battery and other circuits can be used for a very long time.

jskasia, 2020-09-25:
If the electrode (the needle that goes underneath skin) can be cleaned or replaced, the we can re-apply the old sensor. Might need to replace battery from time to time, and reset the circuit. See sample here: https://www.youtube.com/watch?v=Sh6NXaC8EFU&t=893s

cominixo, 2020-10-22:
If anyone's interested, my [https://github.com/cominixo/OpenLibreNFC](app) can modify the Libre's memory, that includes resetting the age value to 0. Sensors usually last 7 days extra and are decently accurate with calibrations.

Spshulem, 2020-12-18:
@cominixo any updates on getting this to work with the Libre 2?

cominixo, 2020-12-18:
Not really, don't have a libre 2 to test with and I probably won't be able to use the same method anyway.

jskasia, 2020-12-19:
@cominixo your app works great. I tested with 2 libre 1 sensors. Both were able to work upto 30 days until they died. With help of the calibration on xDrip. The accuracy maintained thoughout 30 days period. Thanks so much. By the way, where did you get the auth code for different sensors from?

cominixo, 2020-12-19:
Well the password is only one actually, the different IDs were basically from looking up things, for the password I believe you can find it in LibreLink code.

CarlHelberg, 2021-01-30:
Hi @cominixo 

Would it be possible for you to make a branch of this app written in C#?
I think it would be easier for you since you already know how everything works.
Using xamarin it would then be compatible with IOS and Android.
Otherwise, can I give it a shot at rewriting it?

Salva991, 2021-01-31:
Hi, i'm from Italy and nice to meet you all, @cominixo I' ve just installed your app, scan the expired sensor and reset age, but sensor still remain expired, i do wrong? Can you explain me? It is a very cool app 😁👍

cominixo, 2021-01-31:
@CarlHelberg you can rewrite it yeah, would be too much work for me right now. @Salva991 make sure you're not using any official apps to read it, and if you reset the sensor age before ~14d12h when it dies. Also make sure you're actually scanning the sensor again after you press the reset age button.

Salva991, 2021-01-31:
@cominixo so I have to reset before it expires, right? I have tried with the sensor already expired, should I try with the next one?

cominixo, 2021-01-31:
Yeah if it's past that mark you should just do it with the next one, but you can still do it after it offivially expires, just not after the extra 12h or so it has.

Salva991, 2021-02-01:
Om, I do it with yhe next one, I will give you feedback in two weeks 👍👍

TanyaStoman, 2021-02-05:
@CarlHelberg have you had successes with writing the app for IOS?

CarlHelberg, 2021-02-05:
@TanyaStoman not yet, been a bit busy this week, but hoping to look at it this week.
As a side note, on IOS if you have Testflight + Diabox app, they say you can also reset the sensor.
I haven't tried this yet, so can't confirm.

MMW12356, 2021-03-02:
AFAIK, US 14 day and Libre 2 sensors cannot be extended as they are encrypted.

charebl190, 2021-04-07:
> If anyone's interested, my [app](https://github.com/cominixo/OpenLibreNFC) can modify the Libre's memory, that includes resetting the age value to 0. Sensors usually last 7 days extra and are decently accurate with calibrations, just won't work with anything official. Only works with non-US Libre 1 sensors for now.

Hi @

> If anyone's interested, my [app](https://github.com/cominixo/OpenLibreNFC) can modify the Libre's memory, that includes resetting the age value to 0. Sensors usually last 7 days extra and are decently accurate with calibrations, just won't work with anything official. Only works with non-US Libre 1 sensors for now.

Hi @cominix can you send me the app file for the extended life for the freestyle libre app via WhatsApp ot email pls bc i can't manage to build it my sensor expires on Monday if you can help soon😔

glaucoqs, 2021-04-07:
> Hi @
>
>
> Hi @cominix can you send me the app file for the extended life for the freestyle libre app via WhatsApp ot email pls bc i can't manage to build it my sensor expires on Monday if you can help [...]

Hi. You can download and install the app in this link: https://github.com/cominixo/OpenLibreNFC/releases/download/v1.0.1/openlibrenfc-1.0.1.apk
Remember to reset age before expire. (~13,8 days)..
And remember to use the no official app (ex.. xdrip) first days, together with oficial app to calibrate xdrip..
If you have any questions, add my WhatsApp (Brazil) +5548996075675
👍

charebl190, 2021-04-07:
> If anyone's interested, my [app](https://github.com/cominixo/OpenLibreNFC) can modify the Libre's memory, that includes resetting the age value to 0. Sensors usually last 7 days extra and are [...]

@cominixo please if you can send me the app via email bc i can't manage to download it my email is zaarourc45@gmail.com please

glaucoqs, 2021-04-07:
> > If anyone's interested, my [app](https://github.com/cominixo/OpenLibreNFC) can modify the Libre's memory, that includes resetting the age value to 0. Sensors usually last 7 days extra and are decently accurate with calibrations, just won't work with anything official. Only works with non-US Libre 1 sensors for now.
> 
> @cominixo please if you can send me the app via email bc i can't manage to download it my email is [zaarourc45@gmail.com](mailto:zaarourc45@gmail.com) please

Is not possible send the file by e-mail. Google blocks.

charebl190, 2021-04-08:
> If anyone's interested, my [app](https://github.com/cominixo/OpenLibreNFC) can modify the Libre's memory, that includes resetting the age value to 0. Sensors usually last 7 days extra and are [...]

Please send me the apk via mail zaarourc45@gmail.com  pls bc i can't manage yo download it

charebl190, 2021-04-08:
Can someone send it for me by mail? Pls
Zaarourc45@gmail.com i can't manage to build the app and my sensor expires on Monday pls

HimenBluepixel, 2021-07-28:
@cominixo i have a Libre 2 Canadian sensor. I want to activate the sensor with Android custom code. Do you have an idea what could be the byte array or command need to send for activate the sensor. I tested your demo code with Non US libre 1 and its working very well. 

Please let me know if you have any idea for same.

cominixo, 2021-07-29:
I have no idea about internals of libre 2s, don't have them yet

Himen007, 2021-08-23:
@cominixo Do you have an idea about how to activate and get glucose data of the Libre Pro sensor ?

Harry5533, 2021-11-05:
I've been using Libre pro for the past two years and able to extend it for 28 days now. sometimes it breaks before reaching 28. But any day after 14 is of great help. 

Need help with it. DM @ https://www.instagram.com/bornunic/ 

Works only with Libre Pro

maujah, 2021-11-23:
Hello

I have a sensor that finishes its cycle in a few days, unfortunately last weekend I lost the reader where the information of the last three days was left,
Is there a method to retrieve the sensor information, or is there no longer a chance?

Harry5533, 2021-11-29:
hi maujah,

If u have nfc phone u can retrieve it

Thanks

RjayG, 2022-05-23:
> Hi. You can download and install the app in this link: https://github.com/cominixo/OpenLibreNFC/releases/download/v1.0.1/openlibrenfc-1.0.1.apk Remember to reset age before expire. (~13,8 days).. [...]

hi,  i use libre 1 but not familiar with xdrip as i do not use that. do i really need to use that if I reset the sensor? if so, can you teach me how to use it/calibrate? just a quick step by step guide should be ok or if you can point me to a resource to read should be fine as well. thanks!

Harry5533, 2022-05-24:
> > > > If anyone's interested, my [app](https://github.com/cominixo/OpenLibreNFC) can modify the Libre's memory, that includes resetting the age value to 0. Sensors usually last 7 days extra and are decently accurate with calibrations, just won't work with anything official. Only works with non-US Libre 1 sensors for now.
> > > 
> > > 
> > > Hi @
> > > > If anyone's interested, my [app](https://github.com/cominixo/OpenLibreNFC) can modify the Libre's memory, that includes resetting the age value to 0. Sensors usually last 7 days extra and are decently accurate with calibrations, just won't work with anything official. Only works with non-US Libre 1 sensors for now.
> > > 
> > > 
> > > Hi @cominix can you send me the app file for the extended life for the freestyle libre app via WhatsApp ot email pls bc i can't manage to build it my sensor expires on Monday if you can help soon😔
> > 
> > 
> > Hi. You can download and install the app in this link: https://github.com/cominixo/OpenLibreNFC/releases/download/v1.0.1/openlibrenfc-1.0.1.apk Remember to reset age before expire. (~13,8 days).. And remember to use the no official app (ex.. xdrip) first days, together with oficial app to calibrate xdrip.. 👍
> 
> What do You mean reset age before 13.8 days and then use xdrip with sensor ? Or start using xdrip before 13.8 days (to compare to original app), then reset before 13.8 days and then use just xdrip ?

-----------------

@olektee  - reset should be done couple of hours before expiry. thats is 13th day after completion 20 hours of usage is safe. Personally i have tried resetting it even after 3 days and it works good. i use Libre pro from India.

mohdmunaf, 2022-10-07:
@Harry5533 Hi, I wanted to activate Libre Pro sensor through Android/iOS app. It would be great if anyone has code snippet.

foxisoft, 2023-01-23:
Hi,
Do you have a solution to extend Libre2 sensor?
Thanks

cadcoke5, 2023-01-23:
I never found one.

TanyaStoman, 2023-01-23:
If anyone ever do find one please share with us

Harry5533, 2023-01-23:
Hi JoernL/LimiTTer,

This works for libre pro not for other versions.

Thanks
H

evrynet1, 2023-02-15:
@cominixo 

> Yeah if it's past that mark you should just do it with the next one, but you can still do it after it officially expires, just not after the extra 12h or so it has.

If you are up to I would be happy to send you some Libre 3 sensors to have a look and maybe also sponsor you a bit :) to see if you can extend the lifetime of the sensors, let me know, thanks.

sharbel101, 2023-04-15:
has anyone found how to extend the life of freestyle libre 2??

foxisoft, 2023-04-25:
Hi,

Now I'm using a Libre 2 sensor, will expire in 2 days. I will try to 
extend it.

Could you send me the right procedure to do this? Step by step. And the 
right app to do. I've installed on my mobile device (Samsung S10) 
OpenLibreNFC. I don't know if is the right/good app, please clarify my.

About to use Libre 3, I'll be happy to use them to try to restart them. 
You could send me some of them. If you agree this, I'll send you my home 
postal address.

I have had this disease for 7 years and with the help of this type of 
sensors I manage to keep glycosylated hemoglobin around 5 mg/dl

This is my email and phone:

+40727058708

***@***.***

***@***.***

Thank you and good luck

ddranchi, 2023-05-07:
> I've been using Libre pro for the past two years and able to extend it for 28 days now. sometimes it breaks before reaching 28. But any day after 14 is of great help.
> 
> Need help with it. DM @ https://www.instagram.com/bornunic/
> 
> Works only with Libre Pro

HI Harry - does your code also activate the Libre Pro sensor?

alija312, 2023-05-24:
managed to extend libre2( French)with openlibre NFC don't know how long it will last.  I use Juggluco as a reader.  Immediately after the expiration I reset and it works.
[image]

evrynet1, 2023-05-24:
@alija312 how about libre 3? any luck with that?

alija312, 2023-05-24:
> @alija312 how about libre 3? any luck with that?

I don't use libre3.  I do not know.  It's a good combination for me with Libra 2. juggluco and G-Watch.

foxisoft, 2023-05-24:
Hi
How do you use juggluco to restart Libre2? Or what do you use to restart Libre2?

alija312, 2023-05-24:
> Hi
> How do you use juggluco to restart Libre2? Or what do you use to restart Libre2?
> 
> 
> > 
> > @alija312[https://github.com/alija312] how about libre 3? any luck with that?
> > 
> > I don't use libre3. I do not know. It's a good combination for me with Libra 2. juggluco and G-Watch.
i used juggluco to measure.  and extended it with OpenLibreNFC

olektee, 2023-05-24:
> /issues/13 <https://github.com/JoernL/LimiTTer/issues/13>#issuecomment-1560924753],

Could You please do some more explanation step by step how to extend this ?
:)

alija312, 2023-05-24:
at the time he stopped working i scan with openlibre nfc 
Reset age
Scan 
Dump memory
Start
And juggluco scan and he started to working.
I'm surprised too
already working 10h 😁

[19 more replies]