JoernL/LimiTTer #15: Update components

tonichub, 2017-02-03:
HI! I'd love if you can update the list of components , because i can't find the right battery and bluetooth module..  And there is an error on description on module bluetooth its said "if you have Android 4 connect this cable to VCC instead of pin 3 and disable all "for Android 4"  lines in the .ino file"  so .. connect cable for android 4 or android 5 ?  
Thank you!

daniel515, 2017-02-06:
i heard that the adafruit bluefriut nrf8001 LE (bluetooth card) should work but i dont know. 
Hope you keep on trying. please update me if you get any info. Sorry for bad english best regards Daniel

BenY1, 2017-02-18:
Any battery will work, even a non rechargable such as a cr2032 but lipo recharchables are convenient. Shipping lipos is difficult within Europe however Chinese companies are generally very flexible in posting batteries. Try Ali express.
Frankly your as well rooting an Android 4 device to at least Android 5 as later android versions are better at re awaking bluetooth devices from sleep not to mention the energy consumption is less. 
Hardware does not need updating. You need a hm17 or hm 11 module as we know that these are compatible. Arduino pro mini with an ATmega chip 328 / 3.3v / 8 MHz. It hardly matters which type of arduino mini you buy as long as its the above spec. 
Configuring the Arduino IDE just select AT mega 328 and select the serial interface port on your pc

apabari, 2017-08-11:
Just checking in - has anyone found a list of components?  I asked a similar question but haven't heard back yet.