JoernL/LimiTTer #17: LimiTTEr "burn" my sensor after 4 days!

chip83, 2017-03-15:
Good morning,
I have a problem with the LimiTTEr.
It has already "burned" 3 sensors.
After 4 days of use, the sensor stops working and I can not read it with LimiTTEr (which always gives me the same value) can not do that even with the Freestyle reader (which tells me that the sensor is terminated).
The only unusual thing I noticed is that xDrip+ recognize the LimiTTEr as "LimiTTerAT + REST". [image]
Why? You may depend on what? What can I do?

JoernL, 2017-03-17:
Hi! As suggested in our chatroom - change Line 91 to: **delay(500);**
Maybe that your HM module is a bit lazy.

chip83, 2017-09-29:
thanks, with your last version the paring work well and now I see only "LimiTTer", but communication with the sensor happens to be hiccup, it receives data for 5 minutes, then no apparent reason no longer receives data even for 8 hours (I do not know if it will resume because it is disconnected). But often and willingly sending values ​​from the limiter to the sensor does not work. What could it be? I use a BLE HM17 and the solution cube NFC card

Abishek92, 2021-03-01:
Hello, We have tried using your method of reading the values from the sensor on our android application directly. 

The sensor stops working after an hour or two of usage. We are obtaining raw data from the Librepro sensor once every minute.
Please let us know if the same thing happens if you use the librepro with the limitter

Thank you,
Abishek