JoernL/LimiTTer #20: Different Hardware

nigelkeirross, 2017-04-12:
Hi
(Excuse me if this is the wrong place to ask questions). 
I am interested in building one of your LimiTTer for my Libre. I have wondered if you have considered  using either the arduino pro micro rather than the mini?
Also would it be possible to use the Blend Micro  (http://redbearlab.com/blendmicro/) arduino that has built-in BLE, and is 18mm * 42mm, slightly larger than yours, but has BLE built in.
Thanks

JoernL, 2017-04-22:
The pro-micro uses more power than the pro-mini.

mattmholliday, 2017-07-24:
I would also love an alternative to the BM019 as its is very expense to source in Australia.

MityaKLG, 2018-03-19:
Hello! Thank you very much for your product. I would like to clarify whether it is possible to use X-NUCLEO-NFC05A1 http://www.farnell.com/datasheets/2339631.pdf  instead of BM019 ?