JoernL/LimiTTer #23: Xdrip+ does not read any data

MazingerZTTR, 2017-09-26:
Good afternoon, 

First of all i'd like to thank JoernL for sharing this project!
I have built two limitters with exactly the same devices (HM11) explained in pdf instructions but i can't read any data from sensors. Xdrip+ connects correctly by bluetooth as you can see in screenshots but then, nothing happens. I have used the latest arduino code in github but it can't read the sensor.
[image]

[image]

[image]

I also have read this "forum" and restarted the mobile phones (even i have tried with 3 different mobile phones but i had exactly the same issue).

I have checked the wiring twice in both Limitters.

Thank you for your help!

chinflon, 2017-11-08:
Hello @MazingerZTTR,

I am having the same problem, LimiTTer connects to xDrip+ but it gets stuck in "Please wait. 2 readings are needed".

Did you manage to solve it?

Also I found something kind of odd. In the Bluetooth collector status I see 'Glucose data from:' and it gets glucose data once but never updates so even if the Bluetooth connects every 5 minutes glucose data does not update.

@JoernL any idea? Thanks for your shared design by the way, great work!

MazingerZTTR, 2017-11-08:
Hi chinflon, 
I used Lbridge code in arduino (https://github.com/keencave/LBridge) and connected it by serial to read the output. I could see that it wasn't reading anything from NFC and resoldered al connections using new AWG 30 wires. Now is working right.

chinflon, 2017-11-08:
Thank you, I will check that.

chinflon, 2017-11-08:
Anyway I think it should work since every 5 minutes I see the NFC reader green LED flash, so I assume the wiring is ok.

@JoernL 
@MazingerZTTR

chinflon, 2017-11-18:
Thank you very much @MazingerZTTR I did as you suggested and worked perfectly!

cqrendo, 2018-03-21:
@chinflon when you say that you did what @MazingerZTTR says you mean that you change to LBridge or you just resoldered with AWG 30 wires or both?

waltsteinchen, 2018-09-27:
Had the same issue, was bad soldering .. 
Turned on the LED each 5 minutes but finally data were broken ;-(
How to debug ?
A very helpful utility is the app "NRF Connect" (used it on another handy in parallel). This allows you to see the bluetooth packets comming from the Limitter device, ...the data field should contain the value. 
In my case it was 0x00.

Just start the scanner.. wait until your device sends something and the press on RAW.. it should look something like the attached picture. Hope this helps
[image]