JoernL/LimiTTer #27: Ideas to port this to ESP32 ?

waltsteinchen, 2017-12-26:
The ESP32 could replace the ATMega easily and has built in Bluetooth so one module less needed in the end.. Any brain-work already done into that direction ?

renaudrenaud, 2018-01-10:
It could be quite easy to do the job, especially with micro python. Maybe latter, I want to build one as indicated before to think on another version. I also have a 3D printer and plan to design a box.

waltsteinchen, 2018-05-30:
Wow, phantastic! .. can we grab the code somewhere ?

idoroseman, 2018-05-30:
still a work in progress...
see https://bitbucket.org/roseman/freestyle-libre-monitor/src/default/

melanrz, 2018-08-08:
idoroseman,your sketch reset esp32 every reading

idoroseman, 2018-08-10:
hi @melanrz, it goes to deep sleep after each reading to conserve battery power. wakeing up involves a reset. is that what you mean?