JoernL/LimiTTer #29: Libre Pro Sensor

Archana-Balan, 2018-06-06:
Hi, 
I've been working on developing the Limitter device in India. We only have the "Libre Pro" sensors available here which stores complete data of 14 days.
 The sensor stores 15 minute trend data in one buffer and 14 days of historic data in the remaining memory blocks. I'm able to read the 15 minute trend data correctly. However the sensor NFC  tag contains only 256 blocks and can store only 4-5 days of historic data. The remaining data is shown as factory locked. Hence I'm able to read only 4 days of historic data
Any suggestions on how I could read the historic data of remaining 10 days ? 
Thanks !

daclaw2k, 2018-07-09:
Hi Archana-Balan,
I have also tried using the Limitter device with the Freestyle Libre Pro but was unable to get any values at all into xDrip+! How were you able to get current trend data? 
I have been able to get my hands on a few sensors from India so am happy to test anything.
Thanks :)

daclaw2k, 2018-10-03:
> Hi,
> I've been working on developing the Limitter device in India. We only have the "Libre Pro" sensors available here which stores complete data of 14 days.
> The sensor stores 15 minute trend data in [...]

Hi Archana-Balan,

Any updates on how you are going? Did you manage to get this working at all? I'm using Glimp with SW3 at the moment with the Libre Pro, working ok but xDrip+ is a lot better.
Thank you very much :)

amit2411amit, 2018-10-25:
hi archana ji.. i have got the protocol to read sensor n return bytes.. can we communicate

daclaw2k, 2018-10-26:
hi @amit2411amit , I am still using the Libre Pro with sony SW3/Glimp app.
I have also built a Limitter and am happy to help with anything. Would like to get it working with xDrip+!

amit2411amit, 2018-10-27:
@daclaw2k pls connect amit2411/at/gmail/dot/com

daclaw2k, 2019-05-06:
FYI this has now been resolved. I have tested this successfully.

Comments from @keencave
Please find the updated LBridge code V1.1.01 here:
https://drive.google.com/open?id=1RLxzbgu5Y2GPyd2cUklRKI-TU94QK5Fg
Please use the Arduino IDE 1.8.9. The INO need to have the Libre2 library "L2read.zip" installed as a zip library with the IDE. Some minor fixes. Added support for Libre Pro (and Libre H). Any test result welcome. I had only an expired Libre Pro for test.

ipsita1917, 2020-09-09:
Hi.. I am using a Libre pro with miao miao 2 and would like to use xdrip. Will this be possible?? Please help!

ipsita1917, 2020-09-09:
@daclaw2k Hi.. I am using a Libre pro with miao miao 2 and would like to use xdrip. Will this be possible?? Please help!

deepakthakur143, 2021-01-13:
@daclaw2k I am also working on the LibrePro sensor and not able to find the History section offset and limit. Also not sure of how many bytes the total data is. Can you help please ?

daclaw2k, 2021-01-19:
> @daclaw2k Hi.. I am using a Libre pro with miao miao 2 and would like to use xdrip. Will this be possible?? Please help!

Hi @deepakthakur143, sorry I can't help, I am not a programmer and didn't write the original code. @MarakM60 (creator of droplet) helped decode it, I just sent some raw logs through to him. Have now moved over to a dexcom g5 as it was much more stable for me.

jain-shreyansh, 2023-08-05:
How can i read the data from the Abott Freestyle Libre Pro sensor, 
Is there some specific protocol that should be used? 
I identified that ISO15693 is being used, but i tried a few RFID readers for ISo15693 from play store, but that didn't work
it was not able to read the data

jain-shreyansh, 2023-08-05:
> Hi, I've been working on developing the Limitter device in India. We only have the "Libre Pro" sensors available here which stores complete data of 14 days. The sensor stores 15 minute trend data in [...]

Can you share how you were able to read the data?

gui-dos, 2023-08-06:
@jain-shreyansh:

@ MarekM60 ( R.I.P. :-( ) passed me some internal info and even a firmware dump of the Libre Pro (https://github.com/gui-dos/DiaBLE/blob/main/Reversing/Dumps/LibrePro-dump.bin) and I started writing a class to support its special FRAM and extended header layouts in iOS (https://github.com/gui-dos/DiaBLE/blob/main/DiaBLE/LibrePro.swift).

I was reported by some Indian guys that they could also connect to a Libre Pro successfully via a MiaoMiao but I never tested real sample and programmed in Java...