JoernL/LimiTTer #32: ESP32 version of LimiTTer (problem with BLE configuration...)

Philoul, 2019-08-22:
Hello,

I'm trying to make an ESP32 version of LimiTTer freestyle reader, but I can't connect my ESP32 to Xdrip+ (when I scan BT devices on my phone I see my ESP32 (BT name = "LimiTTer") and pair it), but impossible to see my ESP from xdrip+ to received my freestyle raw data (the "Scan Blustooth" don't find any device...

I think that my bluetooth configuration on my ESP32 is not ok but I don't know how to make it work... Can you help me please ?

my bluetooth configuration looks like 

```
in header
#include "BluetoothSerial.h" //Header File for Serial Bluetooth, will be added by default into Arduino
BluetoothSerial ble_Serial; //Object for Bluetooth

in setup
    ble_Serial.begin("LimiTTer"); //Name of your Bluetooth Signal
    for (int i=0; ( (i < MAX_BLE_WAIT) && !(ble_Serial.hasClient()) ); i++)
    {
      delay(1000);
      Serial.print("Waiting for BLE connection ...");
      Serial.println("");
    } 

in Send_Packet function
      ble_Serial.print(packet);
```

In setup, "ble_Serial.hasClient()" remains "false" (wrong use of it?)

When I arrive on the "ble_Serial.print(packet);" instruction (with a "correct xdrip packet"), I have a crash and reboot of my ESP 
xDrip packet: 173647 216 100 17206

"Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled."

I am sure my bluetooth configuration on my ESP32 LimiTTer code is not good but I don't know how to solve it... 

Can you help me please ?

Philoul, 2019-08-30:
Hello,

I have found a solution:

My new Bluetooth configuration for ESP32 is : 

in Header

```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>
#define MAX_BLE_WAIT 60 // Maximum bluetooth re-connect time in seconds 
#define SERVICE_UART_UUID      "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
BLEServer* pServer = NULL;
BLECharacteristic* pTxCharacteristic = NULL;
BLECharacteristic* pRxCharacteristic = NULL;
bool estConnecte = false;
bool etaitConnecte = false;

class EtatServeur : public BLEServerCallbacks 
{
    void onConnect(BLEServer* pServer) 
    {
      estConnecte = true;
    }

    void onDisconnect(BLEServer* pServer) 
    {
      estConnecte = false;
    }
};

class CharacteristicUART : public BLECharacteristicCallbacks 
{
    void onWrite(BLECharacteristic *pCharacteristique) 
    {
      std::string rxValue = pCharacteristique->getValue();
      if (rxValue.length() > 0) 
      {
        Serial.println("*********");
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);
        Serial.println();
        Serial.println("*********");
      }
    }
};

```

Then in Setup : 

```
  BLEDevice::init("LimiTTer");
  //BLEDevice::getAddress(); // Retrieve our own local BD BLEAddress
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new EtatServeur());
  
  BLEService *pServiceUART = pServer->createService(SERVICE_UART_UUID);
  pTxCharacteristic = pServiceUART->createCharacteristic(CHARACTERISTIC_UUID_TX, BLECharacteristic::PROPERTY_NOTIFY);
  // Create a BLE Descriptor : Client Characteristic Configuration (for indications/notifications)
  pTxCharacteristic->addDescriptor(new BLE2902());
  pRxCharacteristic = pServiceUART->createCharacteristic(CHARACTERISTIC_UUID_RX, BLECharacteristic::PROPERTY_WRITE);
  pRxCharacteristic->setCallbacks(new CharacteristicUART());
  
  pServiceUART->start();

  pServer->getAdvertising()->start();
  //BLEAdvertising *pAdvertising = pServer->getAdvertising();
  //pAdvertising->start();
  Serial.println("UART Over BLE start advertising");
  Serial.println("UART Over BLE wait connection");
    for (int i=0; ( (i < MAX_BLE_WAIT) && !estConnecte ); i++)
    {
      delay(1000);
      Serial.println("Waiting for BLE connection ...");
    }
```

and then for sending information (from String packet) : 

```
      int Packet_Size = packet.length() + 1;
      char BlePacket[Packet_Size];
      packet.toCharArray(BlePacket, Packet_Size);
      pTxCharacteristic->setValue(BlePacket);
      pTxCharacteristic->notify();    
```

It's a little more complicated but it works…

acnofsinger, 2019-12-07:
I have a ton of esp32s laying around.  Can you share any more information on your project?  NFC chip? Battery?  Thanks!

Philoul, 2019-12-07:
You can see my project here:
https://github.com/Philoul/LimitTer-ESP32

=> I used this reader for about 1 month, I made it before receiving my order (Miaomiao 2...)

=> My ESP32 ship is [MH-ET LIVE mini kit](http://mh.nodebb.com/topic/8/mh-et-live-minikit-for-esp32)
=> My NFC reader is [BM019 Solution Cubed](http://www.solutions-cubed.com/bm019/)
My batterie is not integrated and has not battery level monitoring (I used external power pack for charging smartphone)

For packaging I used an old Pump Patch Omnipod

[image]
[image]